schreibfaul1/ESP32-TFT-Library-ILI9486 #6: ESP32 TFT-Library -ILI9486 4"

pawelmielnik, 2021-01-09:
I have been tried to use your ESP32-TFT-Library-ILI9486 on
 SKU: 13587
Part Number: 4inch TFT Touch Shield
Brand: Waveshare
UPC: 614961951242
 4" display.
 Unfortunatelly it dosn't work , do you know what changes should i do to run it on 4" display, or do you know any other existig solution.
Pawel,

schreibfaul1, 2021-01-17:
It may be that MISO / MOSI / SCK are not switched through,
Or the SPI frequency is too high, test tft.setFrequency(10000000);

pawelmielnik, 2021-01-18:
Hi,
thanks for support, connection is for sure plugged in. I will try then
with the frequency, now I have 20 000 000.
Paweln

pawelmielnik, 2021-01-19:
Hi,
I checked MISO, MOSI,  CLK ,CS, DC looks ok. I some tests with different settings of CS and DC accordingly changing entries in sketch and cpp.
I changed frequency to 10 000 000  in sketch and in tft.cpp.
Didn't help at all ????
Pawel

schreibfaul1, 2021-01-19:
I am not aware of this particular display. I have one from Waveshare with a different controller that required three bridges to be soldered in. This is this: https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D/blob/master/additional%20info/Waveshare%202.8%20Display.jpg

pawelmielnik, 2021-01-20:
I think that your advice goes in the right  direction.
I'm sending you the web page of this display,
https://www.waveshare.com/4inch-tft-touch-shield.htm . Of Course the
layout is different then what is presented in your picture.
There is a switch ICSP to configure ISP. My understanding is that if my
ESP32 doesn't have ICSP I have to change the position of switches
(opposite to default.
I will test it later today.
Pawel

closed 2021-04-20 by stale[bot]