schreibfaul1/ESP32-TFT-Library-ILI9486 #7: 3.5" TFT LCD with strange pin out

Jordise2002, 2021-03-08:
Hi I bought this tft out of amazon.
I think it was designed for the arduino mega but I was wondering if I could use it with an esp32. But I can't understand how to connect it to the microprocesor.
[image]

katurov, 2021-05-03:
Just take arduino pin-out and use it's map to understand where is power, SPI and so on. I hope this will be easy

optimaxch, 2021-06-05:
Jordi,

.................
// Digital I/O used
//------SPI----------------
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
//------I2S audio----------
#define I2S_DOUT      25
#define I2S_BCLK      26
#define I2S_LRC       32
//------TFT-----------------
#define TFT_CS        15
#define TFT_DC         9

TFT tft;
WiFiMulti wifiMulti;
String ssid =     "2.4GHz";
String password = "2020!Dr3221";
.......................
exactly like schreibfaul code, 
and put 3V to LED_BKL

  opti

closed 2021-08-13 by stale[bot]