schreibfaul1/ESP32-TFT-Library-ILI9486 #14: spi_TFT->writePixels((uint8_t*)colors , len * 2); // Module SPI has function writePixels???

alexismailov2, 2023-12-14:
IMHO Spi should not know anything about pixels...
    spi_TFT->writePixels((uint8_t*)colors , len * 2);

mlekolak22, 2024-09-03:
I have a problem with compiling code, I got this warning:

In file included from C:\Users\Mikolaj\AppData\Local\Temp\.arduinoIDE-unsaved202483-8588-308c57.gbnde\sketch_sep3a\sketch_sep3a.ino:2:
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h: In member function 'void TFT::TFT_DC_HIGH()':
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:238:54: error: 'GPIO' was not declared in this scope
  238 |         inline void TFT_DC_HIGH() {if (TFT_DC < 32) {GPIO.out_w1ts = (1 << TFT_DC);}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:239:54: error: 'GPIO' was not declared in this scope
  239 |                                    else             {GPIO.out1_w1ts.data = (1 << (TFT_DC - 32));}}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h: In member function 'void TFT::TFT_DC_LOW()':
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:240:54: error: 'GPIO' was not declared in this scope
  240 |         inline void TFT_DC_LOW()  {if (TFT_DC < 32) {GPIO.out_w1tc = (1 << TFT_DC);}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:241:54: error: 'GPIO' was not declared in this scope
  241 |                                    else             {GPIO.out1_w1tc.data = (1 << (TFT_DC - 32));}}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h: In member function 'void TFT::TFT_CS_HIGH()':
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:242:54: error: 'GPIO' was not declared in this scope
  242 |         inline void TFT_CS_HIGH() {if (TFT_CS < 32) {GPIO.out_w1ts = (1 << TFT_CS);}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:243:54: error: 'GPIO' was not declared in this scope
  243 |                                    else             {GPIO.out1_w1ts.data = (1 << (TFT_CS - 32));}}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h: In member function 'void TFT::TFT_CS_LOW()':
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:244:54: error: 'GPIO' was not declared in this scope
  244 |         inline void TFT_CS_LOW()  {if (TFT_CS < 32) {GPIO.out_w1tc = (1 << TFT_CS);}
      |                                                      ^~~~
c:\Users\Mikolaj\Documents\Arduino\libraries\ESP32-TFT-Library-ILI9486\src/ili9486.h:245:54: error: 'GPIO' was not declared in this scope
  245 |                                    else             {GPIO.out1_w1tc.data = (1 << (TFT_CS - 32));}}
      |    

Could you tell me what i have to do to get rid of this issue?

alexismailov2, 2024-09-04:
I could not very clearly tell what is the problem but mostly this library depends on GPIO object but looks like it should be less dependent on concrete architecture of microcontroller and be replacable…

Sent from [Proton Mail](https://proton.me/mail/home) for iOS